sparkfun/SparkFun_Qwiic_Scale_NAU7802_Arduino_Library #3: Unable to detect qwiic scale from AZ3166 board with moto:bit qwiic connector.

peterso, 2020-06-19:
### Subject of the issue
Unable to detect qwiic scale from AZ3166 board with moto:bit qwiic connector.

### Your workbench
* What platform are you using?
MS Visual Studio Code

* What version of the device are you using? Is there a firmware version?
Running example1 from this repo.
AZ3166 board library 1.6.5
Qwiic Scale library 1.0.4

* How is the device wired to your platform?
PC =(USB)=> AZ3166 =(edge connector)=> moto:bit =(qwiic 4pin cable)=> qwiic scale =(load cell 4wires)=> 1kg load cell.
* How is everything being powered?
Running on PC USB bus
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue to a gist.
See above connection diagram. Download sketch. https://github.com/sparkfun/SparkFun_Qwiic_Scale_NAU7802_Arduino_Library/blob/master/examples/Example1_BasicReadings/Example1_BasicReadings.ino

### Expected behaviour
Tell us what should happen
Serial monitor should report sensor readings 

### Actual behaviour
Tell us what happens instead
Serial monitor reads:
Qwiic Scale Example
Scale not detected. Please check wiring. Freezing...

nseidle, 2020-06-19:
Can you detect any I2C device using your setup? I'm trying to establish if it's a hardware connection issue or a software issue.

peterso, 2020-06-19:
I currently do not have any other I2C devices but maybe more problematic
ally is that I haven’t found any good references using the AZ3166 Microbit
Edge connector board to Qwiic Interface.

This is where I found the moto:bit which doesn’t have much documentation.

nseidle, 2020-06-19:
The Moto:bit documentation is [here](https://learn.sparkfun.com/tutorials/microbot-kit-experiment-guide/about-the-motobit-board). Do you have a logic analyzer that you could look at the signals to verify the AZ3166 is doing anything on the Qwiic port?

peterso, 2020-06-24:
Thanks for sharing the moto:bit datasheet. I don't have a logic analyzer but I did do some probing with an Arduino Uno and have isolated the problem to be with the NAU7802 chip or library. I have verified my load cell is working and in good order using a HX711 IC with the example library.

Using the I2C Scanner I am able to detect two devices: 0x2A (NAU7802 hard coded address), 0x59 (I assume this is the moto:bit). 

I am still unable to get the NAU7802 library to work with either example1 or example2. Please see a picture of my wiring setup attached. 1st is the desired setup, 2nd is the debug setup with the arduino uno.

https://drive.google.com/file/d/1zmsvTX97v-MbdzgwfJkg1HvIjqW8O3mx/view?usp=sharing
https://drive.google.com/file/d/1o6_NFcKzQaXYCcaUPneg9LqsEZnmdikQ/view?usp=sharing

Any help would be appreciated.

nseidle, 2020-06-24:
What is the output when you run the examples?

peterso, 2020-06-25:
Here is the output on the serial monitor.

Serial Monitor,  Example1: https://drive.google.com/file/d/1wecXNwz5Xigwwbxjc_E7mvNs4_TWDlA6/view?usp=sharing
Serial Plotter, Example1: https://drive.google.com/file/d/1qbl_ckNSS6QSCd0eVBjyM-EmrZSxgEfj/view?usp=sharing

There is no response on the device when the load cell is loaded. I hope I am making a silly wiring error... It seems the scale detected routine is working though. Thoughts?

nseidle, 2020-06-25:
Looks like the IC is starting correctly, library is working as expected, values are coming back. This is not an issue with the library.

Please checkout the hookup guide for the [NAU7802](https://learn.sparkfun.com/tutorials/qwiic-scale-hookup-guide). The most common issues are reversed wires and/or wires that are inserted too far and you clamp onto the insulation rather than the copper.

closed 2020-06-25 by nseidle

peterso, 2020-06-25:
@nseidle the spring clamp seemed to be the issue. There is a very, very thin sweet spot for clamping on the wire. This library is working as advertised on the Arduino Uno. Thank you for your support!

This library however is not working with my original target the AZ3166 board which uses a STM32. Any chance this will be supported in the future?

nseidle, 2020-06-25:
We don't do a lot of STM32 but I welcome a PR if someone wants to take it on.

drphil3d, 2020-10-28:
Same issue with the Sparkfun ESP32 Thing Plus