sparkfun/SparkFun_Qwiic_Scale_NAU7802_Arduino_Library #5: Select Channel 2?

lbussy, 2022-06-21:
### Subject of the issue
Query about using the second channel

### Your workbench
This is generic, I have not yet designed my project

### Steps to reproduce
N/A

### Expected behaviour
I would like to be able to select between CH1 and CH2

### Actual behaviour
Nothing yet!

I can see from the [data sheet](https://www.nuvoton.com/resource-files/NAU7802%20Data%20Sheet%20V1.7.pdf) that there are two channels available on the NAU7802.  I realize your breakout board is specifically designed for simplicity and a single channel.  Nevertheless, your library and the one from Adafruit are the only ones I see available right now, and I do see what appear to be vestiges of CH2 support.

I see what I think is turning on support for the cap on CH2 (for stability I believe) in [`Enable 330pF decoupling cap on chan 2`](https://github.com/sparkfun/SparkFun_Qwiic_Scale_NAU7802_Arduino_Library/blob/688f2559abd4d5a9225665a031a5fa992a03e15a/src/SparkFun_Qwiic_Scale_NAU7802_Arduino_Library.cpp#L63)

I also see that you have a member function [`NAU7802::setChannelSelect()`](https://github.com/sparkfun/SparkFun_Qwiic_Scale_NAU7802_Arduino_Library/blob/688f2559abd4d5a9225665a031a5fa992a03e15a/src/SparkFun_Qwiic_Scale_NAU7802_Arduino_Library.cpp#L158-L165) (currently not used) which allows selecting between channels 1 and 2.

Do I understand that part correctly?  So if I wanted to use two channels (two load cells) I should be able to remove the decoupling cap enable and then select one or the other and read them?

nseidle, 2022-06-21:
You are correct, channel 2 doesn't get much use. However it is covered in the [HG](https://learn.sparkfun.com/tutorials/qwiic-scale-hookup-guide#hardware-overview):

Channel 2
The NAU7802 has a 2nd, optional ADC channel that has a much lower input capacitance (5pF vs 14pF) but is otherwise identical to the 1st channel. By default, this channel has a 330pF capacitor across its inputs. This helps reduce noise on channel one. If you’d like to use the 2nd channel, we recommend removing the capacitor by cutting the CAP jumper. Not sure how to cut a trace? Read [here for more information](https://learn.sparkfun.com/tutorials/how-to-work-with-jumper-pads-and-pcb-traces)!

Channel two is brought out to PTH holes:

[image]

There is a cap crossing B+/B- that can be removed as needed by cutting the jumper:

[image]

[NAU7802::setChannelSelect()](https://github.com/sparkfun/SparkFun_Qwiic_Scale_NAU7802_Arduino_Library/blob/688f2559abd4d5a9225665a031a5fa992a03e15a/src/SparkFun_Qwiic_Scale_NAU7802_Arduino_Library.cpp#L158-L165) will allow you to select between the channels.

closed 2022-06-21 by nseidle

lbussy, 2022-06-21:
Great, thank you so much!  I already have a couple of your boards to test with as well so I'll do some reengineering on those when I get a spare moment.

PaulZC, 2024-02-27:
I know this issue is years old, but #7 contains some proto-code which may help.