sparkfun/SparkFun_Qwiic_Scale_NAU7802_Arduino_Library #7: How to assign gain and calibration to channel 2

MYCAMEL222, 2024-02-26:
I am trying to use both channels of the 7802. I can calibrate and set gain on channel1 ....Works fine.

Is there a way to set the gain of channel 2 and assign different calibration factors to channel2.

I am trying to read 2 load cells. One on each channel.

PaulZC, 2024-02-27:
Hi @MYCAMEL222 ,

Yes, but you will need to do it manually.

If you are using a [SparkFun board](https://www.sparkfun.com/products/15242), open the CAP jumper to isolate the 330pF capacitor on Channel 2 _if needed_. Add 47R resistors on the B+ / B- pins if needed - same as A+ / A-.

[image]

The extra code you need will be something like:

```myScale.clearBit(NAU7802_PGA_PWR_PGA_CAP_EN, NAU7802_PGA_PWR); //Disable 330pF decoupling cap on chan 2```

```myScale.setChannel(NAU7802_CHANNEL_2); //Select Channel 2```

```myScale.calibrateAFE();``` should then perform the Internal calibration on Channel 2 instead of Channel 1. Likewise, ```myScale.getAverage();``` etc. should also read Channel 2.

You can read the Channel 2 offset with ```int32_t offset2 = myScale.get24BitRegister(NAU7802_OCAL2_B2);``` and the Channel 2 gain with ```uint32_t gain2 = myScale.get32BitRegister(NAU7802_GCAL2_B3);``` .

I hope this helps,
Paul

PaulZC, 2024-02-27:
Just to clarify, the library only contains one zeroOffset and one calibrationFactor. You will need to create two for channel 1 and 2 and apply them before calling ```getAverage``` etc..

The steps will be something like:

```
// Zero and calibrate Channel 1
myScale.setChannel(NAU7802_CHANNEL_1);
uint8_t gain1 = NAU7802_GAIN_16;
myScale.setGain(gain1);
myScale.calculateZeroOffset(64);
// Ask the user to apply the calibration weight
myScale.calculateCalibrationFactor(weightOnScale, 64);
int32_t zeroOffset1 = myScale.getZeroOffset();
float calFactor1 = myScale.getCalibrationFactor();
// Zero and calibrate Channel 2
myScale.setChannel(NAU7802_CHANNEL_2);
uint8_t gain2 = NAU7802_GAIN_32;
myScale.setGain(gain2);
myScale.calculateZeroOffset(64);
// Ask the user to apply the calibration weight
myScale.calculateCalibrationFactor(weightOnScale, 64);
int32_t zeroOffset2 = myScale.getZeroOffset();
float calFactor2 = myScale.getCalibrationFactor();

// Then in your loop:
myScale.setChannel(NAU7802_CHANNEL_1);
myScale.setGain(gain1);
myScale.setZeroOffset(zeroOffset1);
myScale.setCalibrationFactor(calFactor1);
int32_t weight1 = myScale.getAverage(64);
myScale.setChannel(NAU7802_CHANNEL_2);
myScale.setZeroOffset(zeroOffset2);
myScale.setCalibrationFactor(calFactor2);
int32_t weight2 = myScale.getAverage(64);
```

closed 2024-02-27 by PaulZC

MYCAMEL222, 2024-02-27:
Thanks a bunch. I understand it better now. I've been able to make progress on both channels, just fighting some EEPROM issues now to save the calibration factors. I'll figure that out.

Thanks for the help.

PaulZC, 2024-02-28:
No problem!

For EEPROM, some platforms need ```EEPROM.begin()``` and ```EEPROM.commit()```. ESP32 needs both. Have a look at [Example2](https://github.com/sparkfun/SparkFun_Qwiic_Scale_NAU7802_Arduino_Library/blob/master/examples/Example2_CompleteScale/Example2_CompleteScale.ino) and search / find "EEPROM".

Best wishes,
Paul